blakadder/templates #1607: Template for Genio downlight 1002742 has a problem

kwagnitz, 2022-11-18:
Using the Blackadder template given:
{"NAME":"GenioDLightCCT","GPIO":[0,0,0,0,0,0,0,0,47,0,37,0,0],"FLAG":0,"BASE":48}
two problems...
1. The colour balance bar goes the wrong way.  Fixed by changing PWM_i to PWM for that pin.
2. With the 'warm' LEDs up, toggling off does not turn 'warm' off.

This template works better:
{"NAME":"GenioDLightCCT","GPIO":[0,0,0,0,0,0,0,0,417,0,416,0,0,0],"FLAG":0,"BASE":11}

blakadder, 2022-11-18:
I don't know what you mean, that is the template that's on the page for that device https://templates.blakadder.com/mirabella_genio_1002742.html

kwagnitz, 2022-11-18:
Yes, sorry.  For the RGB version, you have that template (except for base of 18, which seems to make no difference).
I started with the "CCT" version, ignorant of what that even means.
I did have to remove a capacitor to flash, so I must have the non-CCT version.
The difference still escapes me.  The CCT version isn't dimmable?
Using the same part number for two different lamps is pretty dumb.

BTW, thank you for your efforts, without which I'd be stuffed.