blakadder/templates #1659: DIYMORESOILDHT11 template

ploeffler, 2023-01-12:
...generates 
Analog1 Range
and 
Analog2 Range

which is which - showing no reasonable data

blakadder, 2023-01-13:
can you elaborate your stream of thought?

closed 2023-01-30 by blakadder

ploeffler, 2023-03-21:
hi @blakadder, sorry for the long delay
(I destroyed my device with overvoltage and had to wait for a new one)

ok, I activated the template as suggested and there are definitions for 2 analog-inputs, one on GPIO32 and the other one on GPIO34

my question(s) now is/are:

at which GPIO is the capacitive sensor?
and
what is the second ADC doing?

hope that makes it more clear

regards
peter

blakadder, 2023-03-21:
I wouldn't know because I do not own the device

ploeffler, 2023-03-21:
:(
oh, I thought thats your template definitions -sorry

ploeffler, 2023-03-21:
update: the soilsensor is on GPIO32, GPIO34 is not connected