blakadder/templates #1660: [Site issue] Template wrong - Shelly 2.5

Hoerli1337, 2023-01-12:
I just flashed myself a Shelly 2.5 with Tasmota.
This worked great with the OTA tool here:
[yaourdt/mgos-to-tasmota](https://github.com/yaourdt/mgos-to-tasmota)

Unfortunately, the template cannot be saved.
I should actually insert the following template:
`{"NAME":"Shelly 2.5","GPIO":[56,0,17,0,21,83,0,0,6,82,5,22,156],"FLAG":2,"BASE":18}`

After I clicked save, the Shelly restarts, but then has the following template active:
`{"NAME":"Shelly 2.5","GPIO":[320,0,32,0,224,193,0,0,640,192,608,225,3456,4736],"FLAG":0,"BASE":18}`

Currently I only get the temperature sensors, but no ernegy data.
I can't overwrite the template either, because it always jumps back to the one of the installation tool.
Resetting the device does not help either.

What can I do to get the energy data back?
[image]
[image]

blakadder, 2023-01-13:
the template is saved and corresponds to https://templates.blakadder.com/shelly_25.html

closed 2023-01-13 by blakadder

Hoerli1337, 2023-01-13:
Note for all who come by here:
The Shelly must be without power for 30 minutes after flashing, so that the energy values are displayed.
[comment-1003702107](https://github.com/arendst/Tasmota/issues/5592#issuecomment-1003702107)