blakadder/templates #1721: [Site issue] Suraielec 40A Heavy Duty Outdoor Plug (UBTW01B) comes with unsupported chip

benjinne, 2023-04-05:
https://templates.blakadder.com/suraielec_UBTW01B.html

Suggests the UBTW01B comes with a TYWES3 module, however mine came with a CB3S which doesn't appear to use an ESP32.
Is it possible to swap mine with a TYWES3 or should I just return it?

Here's a picture of what I have

[image]

I tried flashing with tasmotizer, however it wouldn't connect, then I realized the chip was different and most likely not supported.

blakadder, 2023-04-05:
Yes it is possible...https://blakadder.com/replace-tuya-esp12/

benjinne, 2023-04-05:
> Yes it is possible...https://blakadder.com/replace-tuya-esp12/

Awesome! I'll give this a shot.

benjinne, 2023-04-05:
I noticed this flasher only comes with E or F versions. https://www.amazon.com/DORHEA-Firmware-Downloader-Downloade-Compatible/dp/B08P37JQYX

Should I go with that or buy the S version and program it after I swap the chip?

https://www.amazon.com/dp/B0B37MMDKW/?tag=blakaddertemp-20

dirtbikr, 2023-06-10:
I ran into the same thing. Instead of swapping the module for an ESP, I used OpenBeken and flashed the existing module.

I created a teardown with instructions and pictures here: https://www.elektroda.com/rtvforum/viewtopic.php?p=20612154#20612154