blakadder/templates #1734: [Site issue]POWSAV 5+5AC 4USB Power Strip Hardware Changed

technowizard79, 2023-05-02:
The POWSAV 5+5AC 4USB Power Strip has unfortunately had a significant hardware change.  It's now only 4 Smart + 4 AC + 4 USB, and the chipset has changed to the WB3S module.  It'll now need an ESP-12F or ESP-12S (rear pins are not used) transplant for Tasmota.

[image]

blakadder, 2023-05-08:
looks like you're confusing two devices since a change in number of outlets is not just a hardware change

technowizard79, 2023-05-10:
Not really confusing two devices, that's just what Powsav has changed over to now.  Clicking the Amazon link on the template page takes you to the new device, and the older 5+5 plug device doesn't appear to exist anymore.  Perhaps it's splitting hairs, but I thought it worth noting to unsuspecting users, especially since the device that does arrive is now using the incompatible wifi module.  Either way, I did my part saying something... up to you if you think it's something worth noting on the template page, or at least removing the Amazon link to the different device.