blakadder/templates #1740: [Site issue] Shelly Plus Plug S Template

gru-github, 2023-05-10:
Shelly Plus Plug S has four WS2812 LEDs in the outer ring.
This template for both LED Strings (each with two WS2812) on GPIO 25 (left half) and 26 Right half) works for me:
` {"NAME":"Shelly Plus Plug S","GPIO":[0,0,0,0,224,0,32,2720,0,0,0,0,0,0,0,2624,0,0,2656,0,544,1376,1377,0,0,0,0,0,0,4736,0,0,0,0,0,0],"FLAG":0,"BASE":1}
`
Berry scripting in autoexec.be

```
#Index myLeds1 & 2 via Gpio 25,26, Berry index is 0 and 1
myLeds1 = Leds(2,gpio.pin(gpio.WS2812, 0)) 
myLeds2 = Leds(2,gpio.pin(gpio.WS2812, 1))
#Color after boot
myLeds1.set_pixel_color(0,0xff0000)
myLeds1.set_pixel_color(1,0xff00ff)
myLeds2.set_pixel_color(0,0x00ff00)
myLeds2.set_pixel_color(1,0x00ffff)
myLeds1.show()
myLeds2.show()
#method for accessing both strings via RULES
def ledsColor(cmd, idx, payload, payload_json)
  var myColor=int(payload)
  myLeds1.set_pixel_color(0,myColor)
  myLeds1.set_pixel_color(1,myColor)
  myLeds2.set_pixel_color(0,myColor)
  myLeds2.set_pixel_color(1,myColor)
  myLeds1.show()
  myLeds2.show()
  tasmota.resp_cmnd_done()
end
#create new RULES command "LedsColor". Example all Leds to white :
#LedsColor 0xffffff
tasmota.add_cmd('LedsColor', ledsColor)

```

MASHtm, 2023-08-30:
Based on the first example I made this berry script to display relay state on two leds and energy level on the other two
``` 
var power = int(tasmota.get_power()[0])

#Index myLeds1 & 2 via Gpio 25,26, Berry index is 0 and 1
myLeds1 = Leds(2,gpio.pin(gpio.WS2812, 0)) 
myLeds2 = Leds(2,gpio.pin(gpio.WS2812, 1))

def leds_onoff(value, trigger)
  power = value
  if value == 1 
    myLeds1.set_pixel_color(0,0x00CC00)
    myLeds1.set_pixel_color(1,0x00CC00)
  end
  if value == 0 
    myLeds1.set_pixel_color(0,0xCC0000)
    myLeds1.set_pixel_color(1,0xCC0000)
    myLeds2.set_pixel_color(0,0x000000)
    myLeds2.set_pixel_color(1,0x000000)
    myLeds2.show()
  end
  myLeds1.show()
end
tasmota.add_rule("Power1#state", leds_onoff)
tasmota.add_rule("Power1#boot", leds_onoff)

def leds_energy(value, trigger)
  var ledcol = 0
  if power == 1
    var maxpower = 2500.0
    var currentw = maxpower
    if value < maxpower
      currentw = value
    end
    var red = int(currentw * 255.0 / maxpower)
    var green = int(255 - red)
    ledcol = (red << 16) + (green << 8)
  end
  myLeds2.set_pixel_color(0,ledcol)
  myLeds2.set_pixel_color(1,ledcol)
  myLeds2.show()
end
tasmota.add_rule("Energy#Power", leds_energy)
```

As above saved as autoexec.be

bigking94, 2023-09-23:
The second script from MASHtm is great.

But when you turn it off, only one side of the LED ring lights up.
When switched on, the entire ring lights up green Super ^^

Can you adjust the config so that the red LEDs also completely illuminate the ring?

Thank you ^^

MASHtm, 2023-09-24:
One half of "the ring"... basically two of the four LEDs show the power consumption. 0W is green ..orange .. 2500W is red. This is also the reason for the delay because the color is only updated when the event handler is called. 

Since only the other two LEDs show the "power relay state" only those turn red if power is off. 

And it works as designed so far;-)

unreal4u, 2023-12-16:
Hi guys, I have a question since I'm not familiar at all with Tasmota and even less with berry scripts: I wanted to actually flash ESPHome on my Shelly plug plus s because I'm familiar with that but realized there is now a safeboot mode or something like that that will prevent me from running ESPHome after a reboot (?)... or risk to soft-brick the device which probably means having to open it up and flash it with serial anyway.

So I guess I'll just have to stick to Tasmota for now: is there any way to expose the LEDs directly to Home Assistant through berry scripting? What do I want to achieve with this? I want to use the LEDs to show in which tariff I am currently in (low or normal) but following the ton of rules I have already set up for all my other lights in the house (for example by setting brightness depending on either lux values from other sensors or time of day / season of the year, that kind of stuff). 

So: is exposing the LEDs as a controllable LED to HA possible? 

----

For other newcomers: this is the documentation on how to control the LEDs: https://tasmota.github.io/docs/Berry_Addressable-LED/#led-strips-matrix-and-sub-strips and you can control the brightness of the LEDs by passing on a third parameter to `set_pixel_color` which goes from 0 to 100.
There is also a MQTT integration which might be useful: https://tasmota.github.io/docs/Berry/#mqtt-module . I think I might be capable of exposing the entities through there but I really really don't want to learn about all that and having to maintain it when inevitably in a few years time the spec changes.

Parsec43, 2023-12-27:
Hi! Thanks for sharing, but unfortunately I can't work with the two templates. Simply copy/paste does not work, I get the following error message: 

"BRY: Exception> 'value_error' - invalid GPIO number stack traceback:

	<unknown source>: in function `assign_rmt`

	<unknown source>: in function `ctor`

	<unknown source>: in function `init`

	<unknown source>: in function `main`"

Sorry I'm not an expert, but I was just hoping to get the LEDs on my Shelly Plus Plug S to light up without having to spend days learning the programming and syntax... 

What am I doing wrong? I suspect that in the lines 

"#Index myLeds1 & 2 via Gpio 25,26, Berry index is 0 and 1

myLeds1 = Leds(2,gpio.pin(gpio.WS2812, 0)) 

myLeds2 = Leds(2,gpio.pin(gpio.WS2812, 1))" 

I must enter something else. What? An example would be helpful. Only "25" and "26" hasn't worked so far or I've made another syntax error in the process.

And one more question: 

Does the line "{"NAME": "Shelly Plus Plug S", "GPIO":[0,0,0,0,224,0,32,2720,0,0,0,0,0,0,0,2624,0,0,2656,0,544,1376,1377,0,0,0,0,0,0,0,4736,0,0,0,0,0,0], "FLAG":0, "BASE":1}" belong to the template? Do I have to execute it first? Adapted to the name of my device in the network? Do I have to integrate it later in the autoexec.be?

And once it works in the console... As I understand it, I then have to make an "autoexec.be" out of it and place it under "Manage File System" so that it is in the autostart. Right?

gru-github, 2023-12-27:
Shot in the dark: Looks like the template (see first post) is not configured (Settings -> Other Settings -> Template)

Parsec43, 2023-12-27:
Thx, that was it! One have to do this first: https://templates.blakadder.com/howto.html

Loic760, 2024-02-01:
> Based on the first example I made this berry script to display relay state on two leds and energy level on the other two
>
> ```
> var power = int(tasmota.get_power()[0])
>
> #Index myLeds1 & 2 via Gpio [...]

how can I change this script to implement overall brightness control ?

dnlbrg, 2024-02-12:
You can try to append the brightness (as percentage) after the color.

**myLeds1.set_pixel_color(1,0xCC0000*,50*)

Mikecodex, 2024-04-10:
I'm trying to install some sensors in the Shelly Plus plug S. Do you guys know where I can get the real pinout in relation to the template and the pins function? Like a picture pointing to which Gpio is which.

justspacedog, 2024-05-28:
I have to switch the "1" and the "2" next to "WS2812" at GPIO 24 and 25 in order for it to work (to control all 4 LEDs within the Web UI, otherwise only 2 LEDs are controlled). Then the button switches only the light and not the relay, so I have to change the button to Switch and use "SwitchMode 4" with an easy "Rule1 ON Switch1#Action=TOGGLE DO Power1 TOGGLE ENDON".

All this only works as long as the plug is connected to power. When it is unplugged and replugged. Is there anything you have done differently @gru-github?

drakkons, 2024-08-05:
I hope someone answers... even if no one seems to be interested.
On the official page for this device there are two different configurations, on this one there is a third. Which is the correct one? The script reported above does not work or works partially. Can I know if anyone has solved this problem in some way? The idea of ​​showing the status on half of the LEDs and the power consumption on the other half of the LEDs is great! But I repeat, the script does not work!
Has anyone fixed it? It also seems to me that the GPIO configuration is incorrect in some way. Any ideas?

gru-github, 2024-08-06:
Usually it's the missing template or a wrong one.

vturekhanov, 2024-10-17:
It seems that after upgrading from 13.4.0 to 14.3.0, only the LEDs on GPIO 25 still work.

elratt0r, 2024-10-17:
Same problem here with 14.3.0. Neither with GPIO24 as Led1 nor with any other combination it worked.
Also Setting "Scheme 16" to exclude the WS2812 - 1 from being used as LedLink didn't do anything.

(see https://tasmota.github.io/docs/Berry_Addressable-LED/ => Scheme 14,
but because of ESP32 its Scheme 16, see https://tasmota.github.io/docs/Commands/#light)

Found a workaround:
```
Template {"NAME":"Shelly Plus Plug S","GPIO":[0,0,0,0,224,0,32,2720,0,0,0,0,0,0,0,2624,0,0,2656,0,0,1377,1378,0,0,0,0,0,0,4736,0,0,0,0,0,0],"FLAG":0,"BASE":1}
```

Using WS2812_2 and WS2812_3 instead.

So initializing the LED objects becomes:
```
myLeds1 = Leds(2,gpio.pin(gpio.WS2812, 1))
myLeds2 = Leds(2,gpio.pin(gpio.WS2812, 2))
```

My full autoexec.be:
```
var power = int(tasmota.get_power()[0])

# Index myLeds1 & 2 via Gpio 25,26, Berry index is 0 and 1
# => now 1+2 as template is 2 and 3
myLeds1 = Leds(2,gpio.pin(gpio.WS2812, 1))
myLeds2 = Leds(2,gpio.pin(gpio.WS2812, 2))

def leds_onoff(value, trigger)
  power = value
  if value == 1
    myLeds1.set_pixel_color(0,0xFF0000,127)
    myLeds1.set_pixel_color(1,0xFF0000,127)
    myLeds2.set_pixel_color(0,0xFF0000,127)
    myLeds2.set_pixel_color(1,0xFF0000,127)
  end
  if value == 0
    myLeds1.set_pixel_color(0,0x000000)
    myLeds1.set_pixel_color(1,0x000000)
    myLeds2.set_pixel_color(0,0x000000)
    myLeds2.set_pixel_color(1,0x000000)
  end
  myLeds1.show()
  myLeds2.show()
end
tasmota.add_rule("Power1#state", leds_onoff)
tasmota.add_rule("Power1#boot", leds_onoff)
```

This turns on the whole ring @ red, 50% intensity.

Sidenote for the NTC:
```
AdcParam1 33,8200,10000,3350,0
```
gives equal results compared the original firmware (tested in a temperature chamber over a range of 10°C - 50°C to account for equality).

Dizzy1985, 2024-12-02:
Hallo. 
Ich habe einen Plus Plug S mit Tasmota geflasht.
Wie kann ich die "Zweite" Seite des Led-Ring´s per Web Request steuern?
Das Relais z.B klappt per http://192.168.XXX.XXX/cm?cmnd=Power%20Off
Auch der erste LED-Ring lässt sich mit http://192.168.XXX.XXX/cm?cmnd=Color%20FF00000000 steuern. 
Nur beim zweiten weiß ich nicht was benötige!?
Würde es gerne so steuern das sich sowohl Relais als auch die Led-Ringe komplett unabhängig von einander Steuern lassen (Ohne jetzt z.B. Leistungsmessung Rot / Grün). Einfach nur Relais Ein / Aus , Led 1 ein /aus , Led2 ein/aus und das ganze per Web-Request...

geru-github, 2024-12-02:
Ich denke via MQTT ist die beste Lösung. Damit kann man einen Topic abonnieren, der einen Trigger über "ON ..."  verarbeiten lässt. 
Alternativ ohne MQTT könnte ein Kommando via Json ausgelöst werden. Dahinter könnte dann auch ein Berry Skript stehen.
Doku von Commands:

```
Json 	Input any command as valid JSON
{<Tasmota commands>}
example: cmnd/tasmota/json {"HSBColor":"360,100,100","Scheme": 1,"Dimmer": 10,"CT": 220}
```

justspacedog, 2025-01-10:
> Same problem here with 14.3.0. Neither with GPIO24 as Led1 nor with any other combination it worked. Also Setting "Scheme 16" to exclude the WS2812 - 1 from being used as LedLink didn't do [...]

Thank you for your effort, but its the same problem: If it looses power it only lights up half again. Did you found a permanent solution somehow?