blakadder/templates #1762: [Site issue] Template for Refoss P11 Power Monitoring Plug (P11)

florianwgnr, 2023-07-26:
For support with Tasmota flashing and configuring the device visit [Tasmota Discord](https://discord.gg/Ks2Kzd4).
For support with Tuya-Convert please use their [GitHub Issues](https://github.com/ct-Open-Source/tuya-convert/issues).

The combination of the current template and tasmota 12.4+ showed a weird behaviour, where the socket turned immediately "off" after switching the relay to "on". However, after changing _TX GPIO1_ to _CSE7766 TX_ the socket worked without any issues.

Therefore i ask you to change (or add) the following template for the Refoss P11:

`{"NAME":"Refoss-P11","GPIO":[0,3072,0,3104,576,32,0,0,320,224,0,0,0,4736],"FLAG":0,"BASE":1}`

blakadder, 2023-07-26:
currently listed template is directly lifted from a working P11 plug

florianwgnr, 2023-07-26:
My plug arrived yesterday with the current template on the site but didn't work. I had the same issue as some of the newest amazon reviews and was able to solve it by configuring GPIO 1 to TX instead of none. The plug is now working for me with this template.

That's why I thought of adding the template, maybe with a hint like "if the original template did not work, try this one"

blakadder, 2023-07-26:
that makes sense

florianwgnr, 2023-07-27:
As the issue came back today (turning off after turning on) this issue can be closed as won´t do.

closed 2023-07-27 by florianwgnr