blakadder/templates #1764: Option no longer exists - should it be Interlock?

clmcavaney, 2023-08-07:
In this template there are some rules listed, but there is also a `SetOption14` - this option no longer exists in Tasmota.  I think it should be `Interlock 1,2` but I am not 100% sure.

Specific mention:
https://github.com/blakadder/templates/blob/4bbba0cf2b30cebb888d9cd4e758f7f8ef9ecace/_templates/bauhn_ASPU-1019#L15C11-L15C11

Thoughts?

blakadder, 2023-08-14:
I have no experience with the device and I don't see how interlock plays a role here.

It might be SetOption114 1 to decouple the button

sfromis, 2024-01-22:
Change log for [version 6.5.0](https://github.com/arendst/Tasmota/blob/development/CHANGELOG.md#650-20190319) indeed mentions:
> Remove commands SetOption14 and SetOption63 as it has been superseded by command Interlock