blakadder/templates #1819: Shelly Plus Plug S "V2" - Temperature configuration (copy & paste from Shelly Plus 1)

HKEB, 2023-12-11:
Hi,
I bought 4 x Shelly Plus Plug S  and received Shelly Plus Plug S "V2". Flashing worked fine and the Shelly Plug S templates seem to work correct.
BUT the temperature seems to be the same circuit as the one on the Shelly Plus 1 and gives values close to 0 when used indoor.
Given that I'd add a section to
"Shelly Plus Plug S Power Monitoring Plug (PlusPlugS)"
which is a copy and paste from the Shelly Plus 1 Switch but for the Shelly Plus Plug S V2:

"GPIO 33 is an NTC thermistor with a ?10k bridge resistor. Use “adcparam1 2, 10000, 10000, 3350” to get a sensible value (calibration needed!)"

blakadder, 2023-12-12:
each device page has an edit button to add relevant information such as this

closed 2023-12-12 by blakadder

JoakimCh, 2024-05-24:
> each device page has an edit button to add relevant information such as this

Yeah and you closed this without editing it... Hence I got the wrong temperature on my V2 plug and I had to look up this information here! Why close issues not resolved?